GridProtectionAlliance/openHistorian #8: Output Streams

pietropau, 2016-07-04:
why openhistorian 2.0.347 does not handle "concentrator output streams" ?

ritchiecarroll, 2016-07-05:
Technically it can - it just not listed on the Menu since we didn't think it would be commonly used in the historian.

You can easily modify Menu.xml in the C:\Program Files\openHistorian\ installation folder and add the concentrator output item back to the menu:

`      <MenuDataItem Icon="" MenuText="_Concentrator Output Streams" Description="Manage Output Streams" Roles="*" UserControlAssembly="GSF.PhasorProtocols.UI.WPF.dll" UserControlPath="GSF.PhasorProtocols.UI.UserControls.OutputStreamUserControl" />
`

ritchiecarroll, 2016-07-05:
Please move product questions to discussion board: http://discussions.gridprotectionalliance.org/c/gpa-products/openhistorian

closed 2016-07-05 by ritchiecarroll