GridProtectionAlliance/openHistorian #10: Error creating new internal subscription

pietropau, 2016-07-05:
Error when create a new "internal subscription " and click save on "manage device configuration".
Openhistorian version 2.0.347.0
[image]

Thanks,
Pietro

ritchiecarroll, 2016-07-05:
What type of database are you using so we can try to replicate your issue?

pietropau, 2016-07-05:
mysql 5.7 and OS windows server 2008.

StephenCWills, 2016-07-05:
I wasn't able to replicate the issue. Can you provide some more details? Does this happen every time you save, or did it just happen the first time?

pietropau, 2016-07-06:
it just happen when you save for the first time.

StephenCWills, 2016-07-07:
I tried several times with a couple different data sets, but I could not replicate this issue. Because it is difficult to replicate and because it only happens the first time you save the device, I am going to label this as wontfix and close the issue. If you can provide some more detailed information or a sample data set for reproducing the issue, I will try looking into it again.

closed 2016-07-07 by StephenCWills