GridProtectionAlliance/openHistorian #11: Huge Memory usage

pietropau, 2016-07-05:
When you set e new internal subscriber in opehhistorian (version 2.0.347.0) require high memory usage.

[image]

Thanks,
Pietro

ritchiecarroll, 2016-07-05:
How large of a data set are you subscribing to? For example, how many metadata points exist in the openPDC?

pietropau, 2016-07-05:
Openpdc is connected up 216 PMU (200 Voltage, 600 Currents).

Pietro

StephenCWills, 2016-07-05:
I wasn't able to reproduce the issue. Can we get the hardware specs of the system running the openHistorian? Also, do you know about how long the openHistorian was running before you took the screenshot?

pietropau, 2016-07-06:
Hardware specs: Dual-core Opteron 2.40 Hz with 8 GB RAM. 
About memory, it takes two minutes for reach 8 Gb

pietropau, 2016-07-06:
- I reset Openpdc and openhistorian configuration with "configuration setup utility"
- Openpdc configured with 1 PDC with 18 PMU and for receive internal metadata
- Openhistorian configured for receive external metadata from Openpdc

[image]

ritchiecarroll, 2016-07-07:
I guess the best thing we can do at this stage since we are not able to replicate your issue is to get a fresh copy of your log files at system start-up. To do this, stop the openHistorian service and delete (or rename) the following files: StatusLog.txt and ErrorLog.txt. Then restart the openHistorian and send us a copy of these log files a after few minutes. This may help us diagnose the issue you are seeing.

StevenChisholm, 2016-07-07:
The openHistorian was originally designed to have a huge memory buffer for the archive files. After using it in production at OG&E, I've determined that Microsoft since Windows 2008 R2 does a decent job caching the files under the surface as long as they stay open. I'm in the process of reworking how aggressive the historian is at caching files. In the mean time, you can edit your application settings. This can be done by changing "MemoryPoolSize" under the "SystemSettings" in the "openHistorian.exe.config" file to something smaller. I think the value is in GB.

pietropau, 2016-07-08:
Now it's openpdc that require huge memory.

[image]

In my opinion there exists a trouble in the configuration for internal connection between your openPDC and open Historian. Are configured both for receive internal/external metada? is it correct ?

Thanks
Pietro

ritchiecarroll, 2016-07-08:
Can we get a copy of your log files?

pietropau, 2016-07-08:
[Health.txt](https://github.com/GridProtectionAlliance/openHistorian/files/353891/Health.txt)
[StatusLog.txt](https://github.com/GridProtectionAlliance/openHistorian/files/353893/StatusLog.txt)
[ErrorLog.txt](https://github.com/GridProtectionAlliance/openHistorian/files/353892/ErrorLog.txt)

ritchiecarroll, 2016-07-08:
Note that these logs look to be for the historian. Since the openPDC is the one with a large memory pool, can we get those logs. Note that it would also be helpful to see your Status.txt as well - this file is only logged every thirty minutes, but it shows detailed status of all system adapters.

In the historian logs you provided I did seeing some errors, e.g.: 

Failure code received in response to server command "MetaDataRefresh": Failed to transfer meta-data due to exception: Failed to open ADO data connection, verify "ConnectionString" in configuration file: Authentication to host 'localhost' for user 'root' using method 'mysql_native_password' failed with message: Reading from the stream has failed.

Failed to transfer meta-data due to exception: Failed to open ADO data connection, verify "ConnectionString" in configuration file: Authentication to host 'localhost' for user 'root' using method 'mysql_native_password' failed with message: Reading from the stream has failed.

---

However, it looks the system may have recovered from these issues.

Also, are you running on a Virtual Machine?

Thanks,
Ritchie

ritchiecarroll, 2016-07-08:
In the logs you have an IEEE C37.118 PDC style connection called PDC12 and a GEP style subscription called HIS. Can you provide a drawing of your configuration? Here is how I would recommend configuration on this system (based on my current understanding):

[image]

StephenCWills, 2016-07-08:
Hi Pietro,

I think Ritchie has the right idea with the architecture diagram. I see a lot of messages in StatusLog.txt about out-of-sequence statistic measurements. Not only is it a highly unusual message, but it seems to be repeating the same IDs and timestamps while the count of out-of-sequence measurements continuously increases. This suggests you may have a data feedback loop, and that would also explain the memory usage going out of control.

It may be easiest to start over from scratch, making your connections according to the suggested architecture diagram posted by Ritchie.

Thanks,
Stephen

pietropau, 2016-07-08:
We are trying to apply the architecture suggested,  starting over from scratch.
The files provided are from openpdc folder. 
The server is not virtual machine
I change PDC wit single PMU for test.

[StatusLog.txt](https://github.com/GridProtectionAlliance/openHistorian/files/354499/StatusLog.txt)

pietropau, 2016-07-11:
First test

1)  I start over from scratch openpdc and openhistorian, I have configured one PMU into openPDC

[image]

2) Openpdc configured for receive internal/external metadata 

[image]
[image]

3) Memory required by openpdc increasing 

[image]
[image]

 Second Test

1)  I start over from scratch openpdc and openhistorian, I have configured one PMU into openPDC

[image]

2) Openhistorian configured for receive internal metadata 

[image]

3) Memory required by openpdc increasing 

[image]

ritchiecarroll, 2016-07-11:
This is a miss configuration - so sorry for the confusion. The configuration listed is creating a feed-back loop. I will try to detail the steps:

ritchiecarroll, 2016-07-11:
1. From openPDC: Connect to PMU
2. From openHistorian: Connect to openPDC, steps are as follows
  * Select "Inputs / Subscription Based Inputs / Create Internal Subscription":
[image]
  * Fill out info *CRITICAL* - set port to ***6165*** for openPDC:
[image]

That should correct issue - note that you may need to re-rerun Configuration Setup Utility for openHistorian and create a new database. Also: With openHisorian service stopped, I would recommend deleting current "C":\Program Files\openHistorian\Archive" folder to clear out current data.

pietropau, 2016-07-12:
Ok now it's working, thanks

ritchiecarroll, 2016-07-12:
Excellent!!

closed 2016-07-12 by ritchiecarroll