GridProtectionAlliance/openHistorian #12: Error setting output stream

pietropau, 2016-07-07:
when you set output stream isn't possible to insert pmu with device wizard, link seems disabled

openhistorian verison  2.0.347.0 

[image]

ritchiecarroll, 2016-07-07:
Although the openHistorian shares the same base code as other products, it wasn't our intention to use the historian to do PDC style work. Are you trying to make the openHistorian act like a PDC to concentrate device data and send it back out as an IEEE C37.118 stream?

pietropau, 2016-07-07:
I'm working hard trying to find  solution for our system, we tried to apply the architecture suggested:
OPENPDC as PMUs connector concentrator
OPENPDC => OPENHISTORIAN connection
OPENHISTORIAN as historian DB 
OPENPDC to create a virtual PDC to other TSOs in order to redistribute data.

The problem is that  OPENPDC => OPENHISTORIAN connection only with a single PDC with 18 pmu saturate the memory of server. So real problem is: how to activate a correct data exchange between OPENPDC and OPENHISTORIAN ?

Thanks
Pietro

ritchiecarroll, 2016-07-07:
Steven sent you a note on changing a setting for memory utilization (#11) - let us know if that helps your issue. Thanks!

closed 2016-08-01 by StephenCWills