GridProtectionAlliance/openHistorian #13: OPENPDC-OPENHISTORIAN INTERNAL SUBSCRIPTION

cosimopis, 2016-09-02:
Hi,

Internal Subscription in OpenHistorian (which permits to transfer data internally from OpenPDC e OpenHIST) is a one way communication from OpenPDC to OpenHIST and not viceversa? Roughly speaking, is OpenPDC which sends data to OpenHIST and not OpenHIST which asks for data to OpenPDC?

Thank you,

Regards

Cosimo

StephenCWills, 2016-09-02:
Hi Cosimo,

The openHistorian asks for data from openPDC. Stated a little more explicitly, the openHistorian takes the role of the subscriber and the openPDC takes the role of the publisher.

The subscriber is responsible for initiating the connection as the TCP client. From a firewall's perspective, this would be an outgoing TCP connection. Once the connection is established, it is responsible for requesting any metadata, real-time data, and historical data it would like to receive.

The publisher is responsible for listening for connections as the TCP server. From a firewall's perspective, this would be an incoming TCP connection. Once the connection is established, it is responsible for responding to the requests made by the subscriber, either by sending a metadata data set to the subscriber or by providing a real-time data stream to the subscriber.

Currently, the subscriber and publisher are intrinsically tied to their roles as TCP client and TCP server respectively. We've had plans for a while to separate the network protocol from the communications protocol to support other scenarios, but have yet been unable to follow through with those plans.

Thanks,
Stephen

closed 2016-09-02 by StephenCWills

ritchiecarroll, 2016-09-02:
Here's a picture if it helps:
[image]

We realize that this is often not the most desirable mode of operation in that you want to connect "out" of your control center environment. This is a feature we are planning to add at some point, but do not have a timeline on implementing a "reverse" style connection.

Thanks,
Ritchie