GridProtectionAlliance/openHistorian #14: SET SERVER TIME ON WEB MANAGER

cosimopis, 2016-09-15:
Hi,

How Can I set Historian web manager local time local/server time? I noted that while on standard manager the time is aligned with the local for the webmanager is two hours ahaed.
How can I set?

Regards,

Cosimo

StephenCWills, 2016-09-15:
Hi Cosimo,

The local time is the time reported by your browser, which uses the local time zone of the system the browser is running on. The server time is the time reported by the server, which uses the local time zone of the system the openHistorian is running on. To change the time that is displayed, you should only need to change the local time zone on the corresponding system.

Thanks,
Stephen

cosimopis, 2016-09-15:
Thanks Stephen!

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

cosimopis, 2016-09-16:
Hi Stephen,

unfortunately both browser and openHIST are running on the same system. Probably this picture can help:

[image]

As you can see the local time zone (lower left windows server time) is different from the server time and local time reported by the web manager. 

Do you have idea of why?

Thanks

Cosimo

ritchiecarroll, 2016-09-16:
On web browser screen both "server time" and "local time" should be in UTC, the clock on your computer will be in whatever timezone you have selected. The idea is that you can compare the difference of your local clock to that of the server which is more relevant when you are using the web application on another computer besides the server, displaying both timestamps in UTC makes them comparable.

cosimopis, 2016-09-16:
I confirm that on web browser screen both "server time" and "local time" are in UTC. So, isn't there a manner to allign them in your opinion?

ritchiecarroll, 2016-09-16:
You want them to show local time?

cosimopis, 2016-09-16:
Yes exactly, mainly for extraction issues could be better to have timestamps in local time.

ritchiecarroll, 2016-09-16:
I see, we will work on an option to allow the entry and display of timestamps in local time as an option.

Thanks!
Ritchie

cosimopis, 2016-09-16:
Thanks to you!!!!!

ritchiecarroll, 2017-01-24:
Task completed:
https://github.com/GridProtectionAlliance/openHistorian/commit/7a9a7d79ca1c7644286fa9f0b7a67618e13d080d

Will be available in next nightly build.

Thanks,
Ritchie

closed 2017-01-24 by ritchiecarroll