GridProtectionAlliance/openHistorian #15: ARCHIVER FOLDER

cosimopis, 2016-11-21:
Hi guys,

we set openHistorian archiving in the folder in the following:

[image]

Strangely, two folder are created. The fist one with the current year and the second one related to 1970 year O.o.

Do you idea of what is happened?

Kind regards,

Cosimo

StephenCWills, 2016-11-21:
Hi Cosimo,

Is it possible that one of your input devices is reporting timestamps in 1970? It does seem suspiciously close to the Unix epoch, so maybe it's an error code. If you navigate into the folder, the names of the files should give you an idea of the full time range of the data in each file.

Thanks,
Stephen

cosimopis, 2016-11-21:
Hi Stephen,

thank you for your prompty reply as usual. The situation is exactly the one I show you with the following snapshot:

[image]

[image]

Is there a manner via OpenPDC to discover which device is characterized by wrong timestamp?
If I understood what you told me there are two devices with minimum timestamp 1970 and maximum 2036. Correct?

Thanks

StephenCWills, 2016-11-21:
When we were having this problem with some of our devices, we found that there actually aren't any displays in the openPDC that allow you to see the date of the timestamps reported by input devices. We had to use the openHistorian to export data from the time range in question to find which devices reported data during that time. Ritchie has since added a feature to the Trend/Export Measurements page of openHistorian where you can mouse over a value in the "Current Value" column to see the full date and time of the value being displayed.

[image]

Currently, there are still no displays in the openPDC that can be used to display the date of the input devices' timestamps.

cosimopis, 2016-11-21:
Great I will follow your suggestion!

[image]<https://cloud.githubusercontent.com/assets/9203145/20486788/1caadc04-afcf-11e6-94a7-f081ae1c897e.png>

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

closed 2016-11-21 by StephenCWills

cosimopis, 2016-11-22:
Hi Stephen,

I followed your approach, unfortunately it seems that there is no wrong timestamp. 
Is there a manner to automatically discard measurements not aligned with local time in openPDC?
Or, alternatively, if the tiemstamp is wrong, is there a manner to fix the timestamp at the local time?

Please let me know.

Kind regards,

Cosimo

cosimopis, 2016-11-22:
....and additionally. Due to this issue, could the circular archiving fail? I mean that the timestamp misalignment could affect the correct removal of old data? 

Regards

ritchiecarroll, 2016-11-22:
A file with a timestamp of 1970 can occur when anytime when timestamp is received from a PMU that has a bad time - this could occur when a device connected with a bad configuration frame / unlocked clock but has since been corrected. 

How large is the file with the 1970 timestamp - does the file modification time continue to grow? Perhaps this was a temporary issue related to a clock?

Are all your devices ABB's reporting with IEEE 1344? Or do you have a protocol mix that includes IEEE C37.118?

Thanks,
Ritchie

cosimopis, 2016-11-23:
Hi Ritchie,

the situation is this:

[cid:image001.png@01D24568.6D1D5580]

The oldest file has “Date modified” 17/10/2016 10:05. It seems that the cyclical deleting set at 20 days does not work properly.
We have some PMU reporting with IEEE 1344 and other with C37.118 protocol…

Any suggestion to fix?

Kind regards,

Cosimo