GridProtectionAlliance/openHistorian #17: linux support

markrey, 2017-01-01:
Hi,

I am currently working on openHistorian for POC and wanted to check if i can put efforts on creating Linux based deployment. ( with Mono )

Any efforts done in the past?
Which components of openHistorian are purely native to windows?

Thanks

ritchiecarroll, 2017-01-03:
I've have successfully gotten openHistorian to run on Linux using Mono - basically same steps as for openPDC:

- https://gpags.sharepoint.com/TeamSite/_layouts/15/WopiFrame.aspx?guestaccesstoken=ADyQzHPxsfTh9qs4glPelL78SoBA1pTJV1%2fWy6b0ct4%3d&docid=08819043371f24a089e4924e86525dd69&action=view
- https://github.com/GridProtectionAlliance/DeveloperExercises/blob/master/Exercise_openPDC_in_Debian_Jessie.md

Like openPDC, the WPF based manager components only work on Windows because WPF is not fully supported on Linux. Also, the self-hosted Web UI options are not yet working - but you can run the service to archive and retrieve data just fine.

It's a bit of a pain right now to install since you will need to manually run steps that happen in the  install-openPDC.sh shell script but instead of referencing openPDC, you need to reference openHistorian.

Let me know if you have some time to help create an install-openHistorian.sh script - love to have you contribute.

Thanks,
Ritchie

markrey, 2017-01-10:
Thanks Ritchie for your support.
I will be trying it this week and will surely keep you posted. Once successful, i will generate pull request.

neerajpal09, 2017-11-24:
Hi @ritchiecarroll ,

I am also trying to port openHistorian to Linux for the same POC thing using mono. But facing some difficulty and It seems like you already did it. 
Please, can you help me by posting or telling me the steps to compile it for Linux support . I am new to this mono and C# things.
I want to port openHistorian with some fast DB like casandra or MongoDB.

Thanks

closed 2024-11-25 by clackner-gpa