GridProtectionAlliance/openHistorian #21: Initializtion Issue

FerriadWang, 2017-03-06:
Hi,
I was trying to write an action adapter to handle the FNET event. I put the built adapter under the openhistorian V2.2.2.0 directory and I saw it in the openhistorian manager. Then I configured the parameters and click 'initialize'. However, there was no message on the console although i had written the onStatusMessage() in the initialize("xxxx"). I also wrote a log logic to check, but it seemed the logics i wrote in the initialize() were not called. In order to ensure my code was right, I debugged the same code (but different dependencies) for the latest Project Alpha V0.2.107.0. This time, it seemed the initialize() ran well, and so did the publishFrame() logic. I could not figure out where was the problem. Any suggestions?
Best,
Ferriad Wang
FNET, UTK

ritchiecarroll, 2017-03-06:
Could be an issue with the dependencies. I would suggest referencing the GSF assemblies (e.g., GSF.Core.dll and GSF.TimeSeries.dll) that were installed with the openHistorian in your adapter project and rebuilding - this way you are using the same assemblies as the openHistorian that will be using your adapter.

FerriadWang, 2017-03-06:
Richie,
Yes, I was doing the way you suggested. I used the assemblies in the openHistorian V2.2.2.0 to build my adapter and tried to run this adapter on openHistorian V2.2.0.
Ferriad

ritchiecarroll, 2017-03-06:
Perhaps the function is throwing an exception? Have you checked the ErrorLog.txt? Have you tried (1) putting code in a try / catch, or (2) attaching to process with a debug build with a break point in the initialize function? Can I see a copy of your `Initialize` function?

ritchiecarroll, 2017-03-06:
That's a big initialize function. One quick thing jumps out, what is the `m_eventRecordFilePath` set to?

Note that by default the openHistorian gets installed with the `NT SERVICE\openHistorian` local user account and this account has almost zero rights to the system - actually the only rights it has at all are those granted by the installer, e.g., read/write access to the installation folder.

If you folder path is anywhere else on the system, the `NT SERVICE\openHistorian` user will not have needed rights. To give the user rights to another folder you will need to right click on the folder and go to security to edit rights available to the user...

Thanks,
Ritchie

FerriadWang, 2017-03-06:
Ritchie,
The m_eventRecordFilePath is set to the {$installation folder}/event_log.txt. One thing to mention is- we use the same log method in our other adapters, however, they can write log successfully.

ritchiecarroll, 2017-03-06:
The best suggestion I have is to compile the adapter in debug mode, copy the DLL and the associated PDB file into the openHistorian installation folder. Then while running Visual Studio as administrator, attach the the openHistorian process, i.e., from VS menu select "Debug > Attach to Process" and find openHistorian.exe. Once attached but a break point in the `Initialize` function and step through code to see what the issue might be.

FerriadWang, 2017-03-06:
Ritchie, 
the issue has been resolved. I tested the adapter code and there was indeed a bug in the initialization. When I fixed the bug, and the code was running well on the openHistorian V2.2.2.0 Debug mode. Then, I moved it to the openHistorian V2.2.2.0 installation folder and tried again. This time, the console suggested that 'the adapter .dll was used by another program'. Here is the error image:
[image]
Then, I tried to remove the action adapter on openHistorian Manager, restart the openHistorain service, reaname the adapter .dll file, and add the action adapter back to the openHistorian Manager, the issue was revolved.
Not sure if it was related to the naming of the adapters because I updated that .dll file for serveral times and did not change its name.

ritchiecarroll, 2017-03-06:
Great! Glad this is working.

closed 2017-03-06 by ritchiecarroll

FerriadWang, 2017-03-17:
Hi Ritchie,

I am trying to replay the local historian data (V2.2.2.0) to debug our action adapter codes. However the built-in local historian input adapter(openHistorian 2.0 (local)) crashed the openhistorian.exe when I initialized it. It seems it established a connection then the openHistorian was disconnected. The input adapter is named “FNETSREAMER”. Here is the message:
[cid:image003.png@01D29F50.3DB94DC0]

Here is the configuration of the input adapter, the outputMeasurement is a self-defined measurement as ‘Analog Value’.
[cid:image004.png@01D29F4F.2C47EE30]

This happened in every machines where we installed the openhistorian V2.2.2.0. I am not sure if I was doing it right. Your help is appreciated.

Ferriad
FNET