GridProtectionAlliance/openHistorian #23: openhistorian web manager

pietropau, 2017-03-29:
Openhistorian web manager is blocked in load state.
Any advice to resolve the issue ?

[image]

[image]

StephenCWills, 2017-03-29:
Your user account probably has not been assigned any roles under the `Settings > Users` page. You would need to have someone with the `Administrator` role add your user account to the system and assign an appropriate role.

pietropau, 2017-03-30:
My user account is in the administrators group, it's too slow and remains in loading status.

ritchiecarroll, 2017-03-30:
Does another web browser like Chrome have the same issue?

pietropau, 2017-03-30:
explorer and chorme, windows reports this error:
Netlogon has failed an authentication request of account xxxxx in domain PROCESSO. The request timed out before it could be sent to domain controller \\DCPCNC02.processo.prv in domain PROCESSO. This is the first failure. If the problem continues, consolidated events will be logged about every 30 minutes. Please see http://support.microsoft.com/kb/2654097 for more information

ritchiecarroll, 2017-03-30:
Looks like this is an authentication issue - is the remote machine on the same domain as the computer hosting the openHistorian?

pietropau, 2017-03-30:
Yes same domain.

pietropau, 2017-03-30:
i found this error in openhistorian error.log

_

> (Outer Exception)
> Date and Time:         3/29/2017 3:23:45 PM
> Machine Name:          WAECNC02
> Machine IP:            fe80::1569:ddd7:9eb1:b2d8%21
> Machine OS:            Microsoft Windows NT 6.2.9200.0
> 
> Application Domain:    openHistorian.exe
> Assembly Codebase:     C:/Program Files/openHistorian/openHistorian.exe
> Assembly Full Name:    openHistorian, Version=2.0.415.0, Culture=neutral, PublicKeyToken=null
> Assembly Version:      2.0.415.0
> Assembly Build Date:   10/3/2016 12:47:08 AM
> .Net Runtime Version:  4.0.30319.42000
> 
> Exception Source:      System
> Exception Type:        System.Security.Authentication.AuthenticationException
> Exception Message:     A call to SSPI failed, see inner exception.
> Exception Target Site: EndProcessAuthentication
> 
> ---- Stack Trace ----
>    System.Net.Security.NegoState.EndProcessAuthentication(result As IAsyncResult)
>        openHistorian.exe: N 2256837
>    GSF.Communication.TlsServer.ProcessIntegratedSecurityAuthentication(asyncResult As IAsyncResult)
>        openHistorian.exe: N 00199
> 
> 
> (Outer Exception)
> Date and Time:         3/29/2017 3:23:45 PM
> Machine Name:          WAECNC02
> Machine IP:            fe80::1569:ddd7:9eb1:b2d8%21
> Machine OS:            Microsoft Windows NT 6.2.9200.0
> 
> Application Domain:    openHistorian.exe
> Assembly Codebase:     C:/Program Files/openHistorian/openHistorian.exe
> Assembly Full Name:    openHistorian, Version=2.0.415.0, Culture=neutral, PublicKeyToken=null
> Assembly Version:      2.0.415.0
> Assembly Build Date:   10/3/2016 12:47:08 AM
> .Net Runtime Version:  4.0.30319.42000
> 
> Exception Source:      
> Exception Type:        System.Exception
> Exception Message:     Unable to authenticate connection to client [::1]: A call to SSPI failed, see inner exception.
> 
> ---- Stack Trace ----
> 
> 
> [3/29/2017 3:58:27 PM] (Inner Exception)
> (Inner Exception)
> Date and Time:         3/29/2017 3:58:27 PM
> Machine Name:          WAECNC02
> Machine IP:            fe80::1569:ddd7:9eb1:b2d8%21
> Machine OS:            Microsoft Windows NT 6.2.9200.0
> 
> Application Domain:    openHistorian.exe
> Assembly Codebase:     C:/Program Files/openHistorian/openHistorian.exe
> Assembly Full Name:    openHistorian, Version=2.0.415.0, Culture=neutral, PublicKeyToken=null
> Assembly Version:      2.0.415.0
> Assembly Build Date:   10/3/2016 12:47:08 AM
> .Net Runtime Version:  4.0.30319.42000
> 
> Exception Source:      
> Exception Type:        System.ComponentModel.Win32Exception
> Exception Message:     No authority could be contacted for authentication
> 
> ---- Stack Trace ----
> 
> 
> (Outer Exception)
> Date and Time:         3/29/2017 3:58:27 PM
> Machine Name:          WAECNC02
> Machine IP:            fe80::1569:ddd7:9eb1:b2d8%21
> Machine OS:            Microsoft Windows NT 6.2.9200.0
> 
> Application Domain:    openHistorian.exe
> Assembly Codebase:     C:/Program Files/openHistorian/openHistorian.exe
> Assembly Full Name:    openHistorian, Version=2.0.415.0, Culture=neutral, PublicKeyToken=null
> Assembly Version:      2.0.415.0
> Assembly Build Date:   10/3/2016 12:47:08 AM
> .Net Runtime Version:  4.0.30319.42000
> 
> Exception Source:      System
> Exception Type:        System.Security.Authentication.AuthenticationException
> Exception Message:     A call to SSPI failed, see inner exception.
> Exception Target Site: EndProcessAuthentication
> 
> ---- Stack Trace ----
>    System.Net.Security.NegoState.EndProcessAuthentication(result As IAsyncResult)
>        openHistorian.exe: N 2256837
>    GSF.Communication.TlsServer.ProcessIntegratedSecurityAuthentication(asyncResult As IAsyncResult)
>        openHistorian.exe: N 00199
> 
> 
> (Outer Exception)
> Date and Time:         3/29/2017 3:58:27 PM
> Machine Name:          WAECNC02
> Machine IP:            fe80::1569:ddd7:9eb1:b2d8%21
> Machine OS:            Microsoft Windows NT 6.2.9200.0
> 
> Application Domain:    openHistorian.exe
> Assembly Codebase:     C:/Program Files/openHistorian/openHistorian.exe
> Assembly Full Name:    openHistorian, Version=2.0.415.0, Culture=neutral, PublicKeyToken=null
> Assembly Version:      2.0.415.0
> Assembly Build Date:   10/3/2016 12:47:08 AM
> .Net Runtime Version:  4.0.30319.42000
> 
> Exception Source:      
> Exception Type:        System.Exception
> Exception Message:     Unable to authenticate connection to client [::1]: A call to SSPI failed, see inner exception.
> 
> ---- Stack Trace ----

_

ritchiecarroll, 2017-04-03:
Is IPv6 enabled? Maybe we should try restricting access to IPv4...

ritchiecarroll, 2017-04-03:
I found this article that may be related:
http://windowsitpro.com/windows/resolve-performance-problems-associated-authentication-scaling

ritchiecarroll, 2017-04-03:
Note the message in the log `No authority could be contacted for authentication` - here's a troubleshooting guide from Microsoft for diagnosing NTLM authentication issues:
https://blogs.technet.microsoft.com/askpfeplat/2014/01/12/quick-reference-troubleshooting-diagnosing-and-tuning-maxconcurrentapi-issues/

pietropau, 2017-04-12:
resolved, IE configuration issue.

[image]

closed 2017-04-12 by pietropau

ritchiecarroll, 2017-04-12:
By turning off or on?

pietropau, 2017-04-12:
off