GridProtectionAlliance/openHistorian #26: OpenHistorian Action Adapter Unknown Stop

FerriadWang, 2017-05-12:
Hello,

Recently we have seen an unknown stop among our action adapters. we had 4 action adpater instances that using a same .dll file. In the mid April, one of them stopped working for about 8 hours until we re-initialized it, but at the same time the other three adpter instances worked well. Not sure if you know this issue or you have fixed it in the later versions.

Regards,
Ferriad
FNET@UTK

FerriadWang, 2017-05-18:
Hi,

Although this issue persists, I would like to by-pass it by reinitialize the trigger using program. Is there any API I can use to initialize a trigger?
Thanks,
Regards,
Ferriad

ritchiecarroll, 2017-05-19:
You can issue an "initialize <adaptername | ID>" to the service console. Are you wanting to automate this?

StephenCWills, 2017-05-19:
From the command line, something like `echo initialize MY_ADAPTER | "C:\Program Files\openHistorian\openHistorianConsole.exe"` should work.

FerriadWang, 2017-05-19:
Thanks for the reply ritchie & Stephen. Yes, we want to automate this in weekly basis. Is there a way to achieve this?

Ferriad

FerriadWang, 2017-05-19:
Thank you both Ritchie and Stephen, the way works at our side

closed 2017-05-19 by FerriadWang