GridProtectionAlliance/openHistorian #28: Add max archive size setting

arkrohne, 2017-06-23:
Currently, the only way to limit archive size is by the MaximumArchiveDays setting. Having an option to limit the archive to a certain size would simplify storage management considerably.

ritchiecarroll, 2020-06-11:
So there is a setting called `DesiredRemainingSpace` which is a floating point number representing the amount of space to leave on a drive - the value defaults to 500GB. At least for archive directories this creates a limit, however, if all archive directories are full, the system has no choice but to fall back on its working directory for archiving, which usually fills up quickly - when that's full it backs up to memory. 

Anyway, I am agreeing here - this setting is needed...

ritchiecarroll, 2020-06-11:
So the tricky part here is simply trying to figure out where to remove the files from when you have multiple archive output locations, consider the following:
```
Working Directory=D:\OHWorking\;
ArchiveDirectories={E:\OHArchive\; F:\OHArchive\}
```
The `DesiredRemainingSpace` option will automatically move from `E:\` to `F:\` once `E:\` drive space hits the threshold.

Now once `F:\` is full you could start engaging deleting files. Logically you remove "oldest files", so these would be removed from `E:\`.

Now the `E:\` drive will have files older than `F:\` and newer than `F:\`, once space is available.

So `E:\` becomes the new "rolling" archive location and eventually will only have newer files than `F:\` at which point `F:\` would now have the oldest files so space is removed from there.

I think that this strategy would work but it might be confusing for someone looking at multiple archive drive locations in that it would never be "sequential" date-wise.

Would also need to think about "how much" to remove as well as "when".

Thoughts?

closed 2024-11-25 by clackner-gpa