GridProtectionAlliance/openHistorian #31: Issue with TCP package

HeYin1, 2017-10-10:
Currently we are using FDRs to model PMU frames with IEEE 2011 standard. The FDR will send multiple data/config. frame merged within a single TCP package.
However, it seems that OpenHistorian can not separate the data/config. frame successfully. It will just treat it as a bad data and report error.
Is there any latest and stable version to solve this problem?

ritchiecarroll, 2017-10-10:
Which 2011 IEEE standard are you referring to?

ritchiecarroll, 2017-10-10:
We've not noted any issues. Is there an FDR on the internet we could connect to and test?

If so, you can send information in a private e-mail - thanks.

HeYin1, 2017-10-10:
I think it is OK. Would you please provide your server address? So that we can set up the FDR. We can talk this in private e-mail address. My e-mail address is: hyin8@utk.edu
Thanks. 
@ritchiecarroll

closed 2018-10-12 by ritchiecarroll