GridProtectionAlliance/openHistorian #32: How to make a synchrophasor through TCP package

HeYin1, 2017-10-17:
Hi,

I am trying to model a real PMU by using FDR. However, the TCP package from PMU seems to be a special protocol, i.e., synchrophasor, shown in follows,
[image]
When I try to model them, the ones from FDR become TCP ones:
[image]
Is there some special requirement for the PMU TCP package? 
By the way, these packages are captured by wireshark,

Best regards,

He
FNET

StephenCWills, 2017-10-17:
I think the dissector in Wireshark is specifically listening to TCP 4712 and UDP 4713. IIRC, you can manually add a temporary rule to decode as a specific protocol through the context menus in Wireshark. Also, you can change the ports that the dissector is listening to in protocol preferences.

Also, as far as synchrophasor protocols go, I'm pretty sure that Wireshark is only capable of decoding the IEEE C37.118 protocol. If your FDR is using a different synchrophasor protocol, then the dissector will not work properly.

Here's a reference to the Wireshark wiki, if it helps:
https://wiki.wireshark.org/IEEE%20C37.118

HeYin1, 2017-10-17:
Thank you. 
I am using IEEE C37.118 protocol for FDR right now. I try to change the port number to 4712 but it is the same result. 
Moreover, in PMU TCP package, each data frame is checked. How can wireshark know that this is synchrophasor frame? So that I can try to modify the TCP part of FDR.

Best regards,
He
FNET

closed 2018-10-12 by ritchiecarroll