GridProtectionAlliance/openHistorian #38: Grafana Abnormality

FerriadWang, 2018-05-15:
Hi,
I find the built-in Grafana would fail to grab data from the openHistorian. 
I am able to grab the real-time/historian frequency data from the .d2 files and I can see the data is streaming in the historian manager, but I am not able to visualize the data on Grafana. Is there any known issue about this?

ritchiecarroll, 2018-05-16:
Not that I am aware of - could be security related - have you tried restarting the openHistorian service?

FerriadWang, 2018-05-16:
I tried restarting the service but it didn't help. It's weird that I can use the same format FILTER command to visualize the data of another interconnection (just change the value after '=').

ritchiecarroll, 2018-05-16:
Have you tried the latest build:
http://www.gridprotectionalliance.org/nightlybuilds/openHistorian/Beta/openHistorian.Installs.zip

ritchiecarroll, 2018-06-13:
Closing this for now, esp with release of 2.5 - let me know if you have any further issues.

closed 2018-06-13 by ritchiecarroll