GridProtectionAlliance/openHistorian #39: how to migrate data from one installation of openHistorian to another computer for trending/visualization

yasharken, 2018-06-12:
Hi, 

Your response to the following question is greatly appreciated. 

We would like to migrate the required data from one installation of openHistorian to a second computer and be able to trend/visualize the migrated PMU data using Grafana tool on the second computer. 

Basically, we would like to know the required files and the step-by-step procedure to achieve the aforementioned data migration. 

Thank you,

ritchiecarroll, 2018-06-13:
If the configuration is standard, all of the archive files will end up in one folder, e.g.: `C:\Program Files\openHistorian\Archive\`. This complete folder, copied while the openHistorian service is not running, will need to be moved to the other machine - I would suggest a clean drop, i.e., do not merge with any existing destination files.

The other thing you will need to do is copy the configuration, i.e., the database - it will need to be migrated the other machine - without the database you don't have the "meta-data" that describes the points. If it's a SQLite based configuration, this is simple, just copy the file to other machine and use the Configuration Setup Utility to point to the new configuration.

For other database, like SQL Server, you will have to backup the database to a file, copy the exported file to the destination machine, them import the database - a little more complicated.

ritchiecarroll, 2018-10-12:
This work out OK for you? Going to go ahead and close this issue - if you have any more trouble, you can continue to interact / comment on the closed issue.

closed 2018-10-12 by ritchiecarroll