GridProtectionAlliance/openHistorian #42: connecting the openhistorian 2 with PMU

saelwahid, 2018-10-11:
Hi 
I was able to connect to the PMU in SEL-421 relay using the pmu connection tester, but I wasn't able to connect the open historian 2, its giving me this error (connection attempt failed)
would you please show me the right steps to add a new input device, I tried the synchrophasor walk through wizard but nothing changed

ritchiecarroll, 2018-10-12:
(1) If you have successfully connected with PMU Connection Tester, you can save the connection information into a file by selecting "File > Connection > Save..." from the main menu:
[image]

By default this will create a file with an extension of ".PmuConnection" - this is an XML file that contains the connection information you used in PMU Connection Tester to connect to your PMU.

(2) Now, from the "openHistorian Manager" - the desktop application, not the web app - click on the "Input Device Wizard" button on the home page (or navigate to "Inputs > Synchrophasor Device Wizard" from the main menu):
[image]

(3) Click "OK" on the "Welcome to Wizard Walkthrough!" front page and answer "Yes" to the "Have you tested your device's connection settings using the PMU Connection Tester?" question. Also answer "Yes" to "Do you have a connection file from the PMU Connection Tester" question.

(4) You should now be on a screen that says "Please enter the path to the connection file from the PMU Connection Tester." - from here click on the "Browse" button and select the ".PmuConnection" file you created in step (1):
[image]

(5) Click "OK" once the file has been selected and the click "Yes" in response to "Would you like the wizard to attempt to make this connection?". On the next screen click "Request Configuration" to attempt the connection to the device.

(6) Assuming the connection is successful, provide any ancillary requested information on the next one or two screens and click "OK" on each - then on the final wizard screen click "Finish":
[image]

(6) Now clicking "Finish" on the bottom right of the screen will complete adding the device to the openHistorian. You can now navigate to the "Graph Measurements" screen to verify that data is flowing into the historian by clicking "Monitoring > Graph Measurements" from the main menu:
[image]

(7) Additionally, you should now find the data in the openHistorian web application ready for trending and export:
[image]

[image]

ritchiecarroll, 2018-10-12:
In response to:
https://github.com/GridProtectionAlliance/openHistorian/issues/41#issuecomment-429351908

Are both the openHistorian and PMU Connection Tester running on the same machine?

saelwahid, 2018-10-12:
yes they are

ritchiecarroll, 2018-10-12:
Is the PMU Connection Tester running and connected while you are attemptoing to connect to the device in the openHistorian Manager - perhaps the device is limiting the number of allowed connections?

ritchiecarroll, 2018-10-12:
Also, could be a local firewall issue of somekind - you might try disabling the local firewall temporatily or adding the openHistorian service as an exclusion.

saelwahid, 2018-10-12:
yes they are both running at the same time, you mean I should close the pmu connection tester while running the open historian. How can I disable the local firewall??

saelwahid, 2018-10-12:
I closed the pmu connection tester and still the error is there, I manually loaded the configuration file to the open historian and the screean light goes from red to yellow
[image]

saelwahid, 2018-10-12:
does that mean its working??

ritchiecarroll, 2018-10-12:
No

ritchiecarroll, 2018-10-12:
Have you tried the Graph Measurements screen?

saelwahid, 2018-10-12:
I can enable the measurements and monitor them, I will try ti inject values to the relay and see.

saelwahid, 2018-10-12:
I can monitor the voltages and currents with their angles, but why it didn't connected successfully without manually loading the config file

saelwahid, 2018-10-12:
those are snapshots of the open historian screen
[image]

ritchiecarroll, 2018-10-12:
Not really sure either - was it a more complex connection using both TCP and UDP?

Glad it's working - going to go ahead and close issue...

You can still comment here as needed.

Thanks,
Ritchie

closed 2018-10-12 by ritchiecarroll