GridProtectionAlliance/openHistorian #43: API open Historian

esoda, 2018-12-17:
I am currently working on openHistorian for POC and I'd like to send the data stored in openHistorian to a time series database, influxDB.
I can not populate the influxdb database.
Is there any guide that I can follow to fix the problem?
Is there any guide that explains step by step the process of "data transmission" from openHistorian to another time-series database?

StephenCWills, 2019-01-02:
There is an output adapter that was designed to send real-time data to InfluxDB, but there is no step-by-step guide to set it up. My recommendation would be to use the openHistorian Manager to set up a custom output adapter (`Outputs > Manage Custom Outputs`) and select `InfluxDB` from the `Type` dropdown. You would also need to configure the Connection String appropriately.

To move archived data from openHistorian to InfluxDB, you would need to do some custom development. A conversion tool could be built based on HistorianDataWalker, and the InfluxDB adapter's code could be used as a reference to help get the data into InfluxDB.

HistorianDataWalker:
https://github.com/GridProtectionAlliance/HistorianDataWalker

InfluxDB adapter:
https://github.com/GridProtectionAlliance/gsf/blob/master/Source/Libraries/Adapters/InfluxDBAdapters/InfluxDBOutputAdapter.cs

esoda, 2019-02-03:
Hi, thank you very much for your reply.
Unfortunately we tried to follow the procedure you indicated ("Outputs > Manage Custom Outputs) and select InfluxDB from the Type dropdown) but it is not working.

We tried to call the authentication APIs inside the .cs, but it responds with the error "404 not found".
We are working with the 1.7 version of influxDB. Is this compatible? Can you recommend a version that definitely works?

I would like to ask you finally if the open historia + influxdb solution is suitable for real time saves.

ritchiecarroll, 2019-07-24:
Version [2.6](https://github.com/GridProtectionAlliance/openHistorian/releases) now includes an adapter for pushing to Azure Event Hubs that you can try as well.

closed 2019-07-24 by ritchiecarroll