GridProtectionAlliance/openHistorian #46: Data missing on export data from openhistorian webmanager

pietropau, 2019-03-26:
We have  Openhistorian connected to openpdc with Gateway transport (localhost:6165).
When you save data from openhistorian webmanager, for example 2 days (6 phasors), we have this issues:

1. Openhistorian: connection to openpdc interrupted 
2. Openhistorian: Loss data caused by point 1
3. Openpdc: Concentrator output stream interrupted
4. Openpdc: Custom output stream interrupted

See data loss on grafana 
[image]

openhistorian logs
[image]

_[3/26/2019 11:21:13 AM] [WAMS] No data received in 5.0 seconds, restarting connect cycle...
[3/26/2019 11:21:13 AM] [WAMS] Data subscriber command channel connection to publisher was terminated.
[3/26/2019 11:21:14 AM] [SNAPENGINE] Pending Tables V1: 6 V2: 1 V3: 0
[3/26/2019 11:21:17 AM] [WAMS] Attempting connection to tcp://localhost:6165...
[3/26/2019 11:21:17 AM] [WAMS] Attempting command channel connection to publisher...
[3/26/2019 11:21:17 AM] [WAMS] Connection established.
[3/26/2019 11:21:17 AM] [WAMS] Data subscriber command channel connection to publisher was established.
[3/26/2019 11:21:18 AM] [WAMS] TSSC algorithm reset before sequence number: 30933
[3/26/2019 11:21:18 AM] [WAMS] Success code received in response to server command "Subscribe": Client subscribed as compact unsynchronized with 5457 signals._

ritchiecarroll, 2019-03-26:
So openHistorian is connected openPDC for data using GEP and occasionally you see a data drop out. Does it automatically recover or are you manually re-initializing the connection?

Thanks,
Ritchie

pietropau, 2019-03-27:
Yes, the connection was automatically re-initializing.

Thanks
Pietro

ritchiecarroll, 2019-05-22:
This still an issue for you?

ritchiecarroll, 2019-07-24:
Hi Pietro - can you try version [2.6](https://github.com/GridProtectionAlliance/openHistorian/releases) and see if things have improved?

Thanks!
Ritchie

closed 2019-07-24 by ritchiecarroll