GridProtectionAlliance/openHistorian #47: ATMEGA2560 Modbus openHistorian display to Grafana

Maudal78, 2019-05-02:
Dear,
I´ve made modbus program on ATMEGA2560 with TCP , I´send 7 Values and I created a device named ATMEGA2560 with HR0,HR1 ..... HR7 Mappings.
If I connect the device I can see all number in Current Value field
[image]

Now my problem is save those data on database and retrive them from grafana to display.

Could anyone help me find documentation to solve this problem? 

Many thanks
Regards

ritchiecarroll, 2019-05-02:
Did you click on "Save Device"?

Maudal78, 2019-05-02:
good evening, what speed !!
Yes, I´ve clicked on save device and insert name and get no error messages.

Thanks

ritchiecarroll, 2019-05-02:
It could be the saved measurements are not being saved to the historian, i.e., are not "associated" with the historian and hence do not show up.

Can you open the openHistorian Manager, i.e., the desktop application not the web page, and navigate to "Metadata > Measurements" from the menu. From here type in "HR Data" in the text box to the left of the "Search" button then click the "Search" button".

I suspect you will see your Modbus points - select one and check its "Historian" association - a drop-down selection. If the point is not set to archive, simply select the openHistorian archive name from the list - normally "PPA". Do this for all points if needed. This "should" start archinving points - if they don't show up immediately, you can try restarting the system (via "Restart Service" button available from desktop manager and web page).

If the points are already associated with the historian, we may need to check for errors...

Thanks,
Ritchie

ritchiecarroll, 2019-05-02:
Also - a good place for troubleshooting advice is here:
https://discussions.gridprotectionalliance.org/c/gpa-products/openhistorian

Maudal78, 2019-05-02:
I´ll try and give you news
Many thanks
Regards
MD

Maudal78, 2019-05-03:
Dear,
I found the solution but I have to set DataSource PDCSTAT
and TAG name mau_MAU256!PMU:ST1
Now is displayng data and thei are stored.
I don't know what I'm wrong with the settings

Thanks
Regards
MD
[image]

ritchiecarroll, 2019-05-03:
`PDCSTAT` is the statistics historian for the openPDC - not the openHistorian.

If you have added the Modbus connection to openPDC instead of openHistorian, you will need to "connect" openHistorian to the openPDC to receive data.

You can do this using the openHistorian Manager desktop application (not the web page). Open the openHistorian Manager desktop application and navigate to "Inputs > Subscription Based Inputs > Create Internal Subscription" from the main menu.

Type in "PDC" into the `Acronym` and `Name` fields. Enter "localhost" into the `Hostname` field and "6165" into the `Port` field, then click the `Next` button and on the next screen click the `Save` button, which can be found on the bottom right of the screen. It will take a few seconds to synchronize all metadata.

This operation will "connect" openHistorian to the openPDC so that it receives all data, including any future updates.

You can validate that openHistorian is receiving all the openPDC data by navigating to "Monitoring > Graph Measurements" in the openHistorian Manager desktop application and expanding "PDC".

As long as the openHistorian is receiving data from the openPDC, it will be archived in the main openHistorian archive and be available from Grafana.

Hope that helps!
Ritchie

ritchiecarroll, 2019-05-22:
I am assuming this issue is resolved, so I am going to close it - please feel free to repoen if you feel it has been resolved - also, you can always ask support questions for the openHistorian here:

https://discussions.gridprotectionalliance.org/c/gpa-products/openhistorian

closed 2019-05-22 by ritchiecarroll