GridProtectionAlliance/openHistorian #48: Full Documentation for OpenHistorian

ankitshah-it, 2019-09-11:
Dear All,

Required Documentation of OpenHistorian which guides any body how can we use OpenHistorian with screen shot. How can we configure a Devices and how it works, so it gives clarity to User at very much extent. If anybody have such documentation then please share it so i can understand the Flow and How  can we use OpenHistorian in best possible way

ritchiecarroll, 2019-09-11:
As an open source project, the documentation is limited - there is actually much documentation to be found, but it's not well organized - the maintainers are often paid for individual enhancements, but as to yet, no one has had the resources to provide a consistent set of documentation.

To begin examining existing documentation, I would start here:
https://github.com/GridProtectionAlliance/openHistorian/wiki

The installation guide has screen shots to assist with initial configuration:
https://github.com/GridProtectionAlliance/openHistorian/wiki/Installation-Guide

Documentation on adding devices is more sparse - I could could perhaps guide you better if I knew what "type" of devices you would like to add, since there are many. For example, synchrophasors (most all protocols), Modbus, DNP3, and more...

Some use and configuration guides:
https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/Use_and_Configuration_Guides.md

ritchiecarroll, 2019-09-11:
Also, documentation for the related openPDC project may provide much insight:
https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/openPDC_Documentation_Home.md

closed 2019-09-11 by ritchiecarroll