GridProtectionAlliance/openHistorian #53: Grafana and OH User Synchronization requires log-on

clackner-gpa, 2020-01-14:
Adding a user to the OH and running /grafana/synchUsers won't result in the user being synchronized to Grafana. 
However if the user account logs into Grafana for the first time it will automatically be synched from the OH.

ritchiecarroll, 2020-01-15:
I first suspected this may have been related to a user having no roles, but currently unless a user actually logs in to openHistorian, they won't be synchronized to Grafana. This seems logical enough, but makes assigning rights in Grafana a process that can't be completed without user logging in. Perhaps an additional flag to the `syncusers` command to force synchronization would be sufficient since this is an edge case.

ritchiecarroll, 2020-02-28:
Fixed #5fb584f16beef4263f3188983cd81d88d5497b6e

closed 2020-02-28 by ritchiecarroll