GridProtectionAlliance/openHistorian #55: Adding New Device (hardware PMU)

arz-ece, 2020-02-04:
Hello Everyone,

I am trying to ADD DEVICE in openHistorian 2.0 portal, so that I can retrieve data seamlessly instead of using the conventional Historian Playback Utility software. 

The problem I have is that **it does not allow me to Save Device (not clickable)**, even after I assign all the Phase quantities in red in picture below! 

The connection file is for a hardware PMU that has been tested with PMU tester and is in use with OpenPDC2.7. I appreciate any help on this.  

[image]

Thanks,
Ali

ritchiecarroll, 2020-02-04:
So long as there are errors on the screen, you will not be able to click "Save". You will need set the "Phase" to a valid value for each the voltages and currents.

BTW, we have improved the algorithm to "guess" the phase based on common PMU phasor labels and this will be in the next openHistorian release coming out soon.

arz-ece, 2020-02-04:
Thank you Ritchie for the information, but I had put all phases to a valid value before and the problem still prevailed.

ritchiecarroll, 2020-02-04:
If there is no "red errors" then the "Save" button should be enabled. You can also press "F12" on the browser to see if there are any reported errors.

arz-ece, 2020-02-04:
I ran this on Internet Explorer instead of Chrome browser and this is the Error I got:

[image]

ritchiecarroll, 2020-02-04:
I think on IE that may be a normal message. Did Chrome report any messages?

arz-ece, 2020-02-04:
> I think on IE that may be a normal message. Did Chrome report any messages?

Chrome did not report any error messages.

ritchiecarroll, 2020-02-04:
Since this new web-based UI for adding devices is giving you trouble, I suggest using the "openHistorian Manager" _application_ than can be found on the start menu.

Just use the new device wizard to add a new device.

Thanks,
Ritchie

arz-ece, 2020-02-05:
Thank you for the prompt reply Ritchie. I have the following problem with the OpenHistorian Manager. I had even stopped OpenPDC and tried to add device, but still same error!

[image]

ritchiecarroll, 2020-02-05:
I though the web page just wouldn't allow you to press the `Save` button, did you see this same error message on the web interface?

Anyway, this message says to cannot connect to device. Does PMU Connection Tester connect to target device OK? You can download tool from here:

https://github.com/GridProtectionAlliance/PMUConnectionTester/releases

Thanks,
Ritchie

ritchiecarroll, 2020-02-05:
FYI - I am closing this issue as this seems like more of a usability issue than a software problem.

Please feel continue this conversation at our discussions board for assistance with software usage:

https://discussions.gridprotectionalliance.org/c/gpa-products/openhistorian

Thanks!
Ritchie

closed 2020-02-05 by ritchiecarroll