GridProtectionAlliance/openHistorian #70: Reinstall under different service credentials does not overwrite HTTP reservations

ritchiecarroll, 2021-12-15:
Further testing shows that original reservations associated with initial service install are not always being removed.

Could be an issue with code call from uninstall operation - perhaps DLL gets removed before call executes reservation removal?

StephenCWills, 2021-12-15:
Removal occurs during install, right before adding new ones.
https://github.com/GridProtectionAlliance/gsf/blob/fb778cb5c3720775d1329222ffbc24306fe6ca46/Source/Libraries/GSF.InstallerActions/CustomActions.cs#L527-L533

StephenCWills, 2021-12-15:
The issue is that it's missing the slash at the end of the URL.
https://github.com/GridProtectionAlliance/gsf/blob/fb778cb5c3720775d1329222ffbc24306fe6ca46/Source/Libraries/GSF.InstallerActions/CustomActions.cs#L919