GridProtectionAlliance/openHistorian #71: How to disable openHistorian debug host automated start

gitgaojiasheng, 2022-02-08:
When I try to debug the openHistorianManager project in visual studio, the openHistorian  will started before openHistorianManager. How to disable this?

ritchiecarroll, 2022-02-08:
As I understand this, because the primary application, openHistorian, is marked as a web application by Visual Studio it's always going to try to start openHistorian first when debugging. 

The simplest way I know to prevent this is to simply "unload" the openHistorian project. For example, right-click on the openHistorian project from within the solution and select "Unload Project" from the menu.

Hope that helps!

Thanks,
Ritchie

StephenCWills, 2022-02-08:
With the openHistorian project selected in Solution Explorer, use F4 to access the Properties menu and turn off `Always Start When Debugging`.

[image]

ritchiecarroll, 2022-02-08:
Ah, the best answer! :-)

GitTurboy, 2022-02-09:
Thank you Very Much! This is helpfull!
> With the openHistorian project selected in Solution Explorer, use F4 to access the Properties menu and turn off `Always Start When Debugging`.
> 
> [image]