GridProtectionAlliance/openHistorian #73: lossless data compression

deathfromheaven, 2022-03-03:
Hello，I want to know the name of the lossless data compression you use in the OpenHistorian and how to find it.

ritchiecarroll, 2022-03-03:
Technically the openHistorian, being based on [SnapDB](https://gridprotectionalliance.org/technology.asp#SnapDB), can use any compression algorithm.

The default one is designed to do a good job with a variety of streaming time-series data sources, we casually refer to this as Time-Series Special Compression (or TSSC) - but there are variations of this, e.g., the one used in [STTP](https://github.com/sttp/). Most of the compression algorithms were developed by @StevenChisholm.

For the the current implementation in openHistorian, you can find the code here, check the `Encode` and `Decode` methods:

* C#: [HistorianStreamEncoding.cs](https://github.com/GridProtectionAlliance/openHistorian/blob/master/Source/Libraries/openHistorian.Core/Snap/Encoding/HistorianStreamEncoding.cs)
* Python: [historianKeyValueEncoder.py](https://github.com/GridProtectionAlliance/openhistorian-python/blob/main/src/openHistorian/historianKeyValueEncoder.py)

deathfromheaven, 2022-03-24:
Thank you very much！I got it.

closed 2022-04-13 by deathfromheaven