GridProtectionAlliance/openHistorian #74: How to get openPDC data in openHistorian.

deathfromheaven, 2022-04-13:
Excuse me，I would appreciate that if you can tell me how to get openPDC data in openHistorian. In other words, how can I transmit the openPDC data to the openHistorian.

ritchiecarroll, 2022-04-13:
Just use an STTP subscription to directly connect to you openPDC from openHistorian to start immediately archiving data. To do this:

1) Open the openHistorian Manager (UI tool from Start Menu - not web UI)
2) Go to `Inputs > Subscription Based Inputs > Create Internal Subscription` from menu and enter following field info:
   1) Acronym: `OPENPDC`
   2) Name: `openPDC System`
   3) Hostname: `IP` or DNS name of openPDC (or `localhost` if installed on same system)
   4) Port: `7165`
   5) Select: `STTP`
   6) Uncheck `Use Source Prefix "OPENPDC!"`
   7) Click `Next` then click `Save`
3) Wait a few seconds, then go to `Monitoring > Graph Measurements` from the menu
4) Data will be successfully archived into openHistorian

deathfromheaven, 2022-04-13:
OK,Thank you very much！

closed 2022-04-13 by deathfromheaven